adafruit/Adafruit_CircuitPython_Bundle #2: is31fl3731 is included but appears to be MicroPython version?

BorisKourt, 2017-02-28:
Hi,

I installed the bundle on a M0 board but I am seeing that `is31fl3731` library tries to import `machine` which is part of the old API. 

Is this a simple fix or a longer term issue? 

I can do a pull request if you point me in the right direction to get started.

Best,
Boris

tannewt, 2017-02-28:
Its just an oversight. We haven't ported the is31fl3731 driver to CircuitPython yet. @tdicola did the other display ports so he can probably give the best pointers.

The core of that code is here if you want to try and follow its pattern: https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display/blob/master/adafruit_rgb_display/rgb.py

BorisKourt, 2017-03-01:
Sorry it looks like `utime` is the problem. I believe it was renamed to just `time`?  
I'll edit it and try.

tannewt, 2017-03-01:
It was switched back to `time`. `machine` should still work on the ESP8266 but on the M0 its been replaced by `nativeio`. Thanks for taking a look at it!

BorisKourt, 2017-03-01:
After switching to `time` the import worked. Unfortunately when I try to:

```python
display = is31fl3731.Matrix(i2c)
```

I get

```sh
>>> display = is31fl3731.Matrix(i2c)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "libraries/drivers/is31fl3731/is31fl3731.py", line 35, in __init__
  File "libraries/drivers/is31fl3731/is31fl3731.py", line 62, in reset
  File "libraries/drivers/is31fl3731/is31fl3731.py", line 67, in sleep
  File "libraries/drivers/is31fl3731/is31fl3731.py", line 44, in _register
  File "libraries/drivers/is31fl3731/is31fl3731.py", line 41, in _bank
AttributeError: 'I2C' object has no attribute 'writeto_mem'
```

Is this another renamed function? 

Best,
Boris

tannewt, 2017-03-02:
I think [these](https://circuitpython.readthedocs.io/projects/register/en/latest/) are the best docs I've written for the drivers stuff. You can ignore the register part for now if you like. It covers bus device which I recommend you use to do the locking. Its there to make sure you keep exclusive rights to the bus as you are using it. (threads and internal code can run at the same time as your code.)

`try_lock` is used to attempt to grab a lock and return whether it succeeds or not. So, you can just do:

```
while not i2c.try_lock():
    pass
```

It will wait until it can grab the lock then. Just make sure that you always call `unlock` after. If you don't, then other code may wait indefinitely.

BorisKourt, 2017-03-02:
I notice that in the doc here:

```python
def __set__(self, obj, value):
    with obj.i2c_device:
        obj.i2c_device.writeto(self.buffer, end=1, stop=False)
        obj.i2c_device.readfrom_into(self.buffer, start=1)
        if value:
            self.buffer[1] |= self.bit_mask
        else:
            self.buffer[1] &= ~self.bit_mask
        obj.i2c_device.writeto(self.buffer)
```

No `try_lock()` appears to be needed.

In the `is31fl3731` library is it a matter of adopting the new syntax (`writeto` instead of `writeto_mem` etc.)? 

Sorry for so many questions.

tannewt, 2017-03-02:
I'm happy to answer questions. :-)

The try_lock is handled by the I2CBusDevice (`obj.i2c_device`). When it enters the `with` statement it claims the lock and then releases it after its done.

The BusDevice also handles the I2C address. (We should really rename the functions to `read_into` and `write`.)

Its a little more work but the bus device definitely makes it nicer. Heres another example with bus device: https://circuitpython.readthedocs.io/en/latest/docs/design_guide.html#i2c-example

closed 2017-04-19 by tannewt

BorisKourt, 2017-04-19:
Thanks for that change. I ran out of time to keep hacking.

commit 2ee99af3 (2017-04-19): maintenance
* Update CircuitPython
* Remove max7219 because its rarely used and written for MicroPython.
* Replace IS31FL3731 driver with one written for CircuitPython. Fixes #2