adafruit/Adafruit_CircuitPython_Bundle #3: Include Examples

tannewt, 2017-04-11:
Ship examples in the bundle like Arduino has examples.

tannewt, 2018-01-24:
This would be awesome to do after all of our libs have examples. Should they be in the same zip or should we ship a separate zip? We'll likely need to change the circuitpython-build-tools to do it.

kattni, 2018-09-30:
We're doing this! Closing!

tannewt, 2018-10-01:
No we're not. The zip for the bundle doesn't include the examples.

kattni, 2018-10-01:
Oh I was thinking this was referring to including examples in the
libraries. Understood.

sommersoft, 2018-10-06:
I was poking around in this a little during some down/distraction time. I missed this comment when I first started looking, and eventually was asking myself where to put the examples in the zip:

> Should they be in the same zip or should we ship a separate zip?

While thinking of pros/cons, I think the answer boils down to expected use cases.

1. **Bundled With Source**: Use Case - On PC/RPi
  - Including the examples inside the `.py` bundle makes it a nice one-stop-shop with `source + examples`. Definitely makes it easier to read different examples and source code, as opposed to using GitHub.

2. **Separate Zip**: Use Case - On MCU
  - I like the thought of allowing a user to `import examples.xxx_simpletest`, from the REPL or otherwise. Allows quick testing of hardware, teaches the modularity of Python modules, or just quickly running a Learn Guide (that uses a simpletest). For this use case though, having the `.py + examples` bundled creates more problems with storage, memory, and import conflicts.

Option 2 is where I lean...

kattni, 2018-10-06:
I agree it comes down to use cases. 

I may be off here, but it seems like the only MCUs you'd be using the `.py` bundle on are the M4+ anyway, which I believe are unlikely (at the moment anyway) to run into storage/memory issues even with the examples included. So we may want to consider both. Include the examples in the `.py` bundle with the already-existing caveat that the bundle is too much for certain boards (and provide the `import examples.xxx_simpletest` experience where possible). Then also ship a separate zip for those who wish to download the examples to have them locally and potentially drag them individually to their board as they go.

sommersoft, 2018-10-07:
> So we may want to consider both.

Haha. I may have been a little too focused on one or the other. I don't think Travis or GitHub will have storage problems... :laughing:

sommersoft, 2018-10-11:
> It can be in the mpy versions as well but contain py files. We teach folks to only drag lib over so its ok if it doesn't have the examples.

👍 I was on the fence about including them in mpy, and definitely wasn't going to mpy the examples if they were included.

>Bonus points for changing the zip so it has a `<name>-<version>` folder with examples and lib in it. That way the lib folders won't get renamed.

I'm not fully sure what you mean here. The best I can come up with is that you want to drop the date stamp from the end of the filename? If that's not it, could you please proved a tree outlining how you want it setup?

tannewt, 2018-10-11:
Sure! Here is what we have now:

```
adafruit-circuitpython-bundle-3.x-mpy-20181011.zip
    lib
        <mpy files>
```

Instead we'd have:
```
adafruit-circuitpython-bundle-3.x-mpy-20181011.zip
    adafruit-circuitpython-bundle-3.x-mpy-20181011
        examples
            <py files>
        lib
            <mpy files>
```

sommersoft, 2018-10-23:
Forgot to close this.. Fixed by [#18](https://github.com/adafruit/circuitpython-build-tools/pull/18) & [#19](https://github.com/adafruit/circuitpython-build-tools/pull/19).

closed 2018-10-23 by sommersoft