adafruit/Adafruit_CircuitPython_Bundle #5: Should MAX7219 be supported in CircuitPython

mrmcwethy, 2017-08-12:
I have been working on at an implementation for max7219 spi support that is compatible with CircuitPython based the oled driver.  I see the micropthon version of max7219 and documentation in the CircuitPython API document for it, but no code.  Did you decide not to implement it?  Should may driver make more sense in the Community Bundle?  I need some direction.

tannewt, 2017-08-12:
Its not intentional. I just haven't gotten to it. We do sell the chip ourselves so I'd be interested in having it under our official support.

The code for those docs is here: https://github.com/adafruit/micropython-adafruit-max7219/blob/master/max7219.py

How about you modify that code to work on CircuitPython and I'll create a separate adafruit repo where it can live? We can also add it to the official bundle then.

mrmcwethy, 2017-08-12:
I like that.  I have a working version for Matrix 8x8.  Also I was thinking that a major feature of the MAX7219 chip is the 8 digit 7-segment LED driver.  I was thinking we could add the API for that as well.
Mike

tannewt, 2017-09-05:
Fixed by 8881d5a16d31d2df9d671887574e041e0fde6e37

closed 2017-09-05 by tannewt