adafruit/Adafruit_CircuitPython_Bundle #6: Cannot build bundle, missing make process.

mrmcwethy, 2017-08-14:
Need help  concerning building the CircuitPython_Bundle.  I have been able to get the source code from GitHub.  When I run build-bundle.py, the process fails to find/execute ["make"].  Where should that program be and where should I install make from?

C:\Users\Mike\Documents\CircuitPlayground\source\Adafruit\Adafruit_CircuitPython_Bundle>python -V
Python 3.5.1

C:\Users\Mike\Documents\CircuitPlayground\source\Adafruit\Adafruit_CircuitPython_Bundle>python build-bundle.py run
Traceback (most recent call last):
  File "build-bundle.py", line 33, in <module>
    make = subprocess.run(["make"])
  File "C:\Users\Mike\AppData\Local\Programs\Python\Python35-32\lib\subprocess.py", line 696, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\Mike\AppData\Local\Programs\Python\Python35-32\lib\subprocess.py", line 950, in init
    restore_signals, start_new_session)
  File "C:\Users\Mike\AppData\Local\Programs\Python\Python35-32\lib\subprocess.py", line 1220, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified

C:\Users\Mike\Documents\CircuitPlayground\source\Adafruit\Adafruit_CircuitPython_Bundle>

mrmcwethy, 2017-08-14:
In the Vagrant VM I built a Python 3.6 version of python from source.  Then install it.  (I needed get a version of zlib to install properly)
Once i had Python installed, I ran python3 build-bundle.py run.  Seems to work.  We can close this issue.

closed 2017-08-14 by mrmcwethy

tannewt, 2017-08-14:
Glad you figured it out!