adafruit/Adafruit_CircuitPython_Bundle #12: Documentation: MAX7219 references micropython class; should be the new adafruit one

mrmcwethy, 2017-09-07:
We need to create a ReadTheDocs file for adafruit max7219 and update circuitpython/index.rst to reference it.  One file we could use is kept in https://github.com/adafruit/Adafruit_CircuitPython_MAX7219/blob/master/README.rst

tannewt, 2017-09-07:
I've created a new ReadTheDocs for it: http://circuitpython.readthedocs.io/projects/max7219/en/latest/

Would you mind adding the link to the CircuitPython docs?

mrmcwethy, 2017-09-07:
>  I've created a new ReadTheDocs for it: http://circuitpython.readthedocs.io/projects/max7219/en/latest/Would you mind adding the link to the CircuitPython docs?—

Sure no problem
Mike

tannewt, 2017-09-15:
This was done in https://github.com/adafruit/circuitpython/commit/dd72fe694526d2b468e189b70ae732f7568fe6e4

closed 2017-09-15 by tannewt