adafruit/Adafruit_CircuitPython_Bundle #20: Default 12c addresses for pca9685.py and motor.py are different.

notthatbrad, 2017-09-24:
Not sure if this is an issue, or just an oddity. the default address of the pca9685 breakout is 0x40. This is reflected in pca9685.py. The learning system documents also show that is the default for the featherwing (https://learn.adafruit.com/adafruit-8-channel-pwm-or-servo-featherwing/pinouts). 

In motor.py the code is trying to set up the motors up on 0x60. This was throwing an error for me. I'm not sure if this is a bug or just something I'm looking at wrong, but thought I would mention it just in case.

mrmcwethy, 2017-09-24:
The adafruit_pca9685.motor.DCMotor class AND adafruit_pca9685stepper.Steppers() both show 0x60 as the default address. I think they both should be 0x40 since the Adafruit board has a default address of 0x40.

tannewt, 2017-09-25:
Thanks for finding this @notthatbrad and @mrmcwethy! Would either of you be willing to submit a pull request to fix it?

mrmcwethy, 2017-09-25:
@notthatbrad and @tannewt .  i have the changes in my area ready to pull req.  I have tested the changes as far as i can, which is putting the new code onto a Metro M0 Express board and writing a test program to creates a instance of DCMotors and Steppers and then seeing it fail looking for an I2C device on address 0x40.  Since I do not have the hardware, i can not show a working case.  If you are comfortable with this level of testing, i can do a pull req. Branch chgaddr60 in https://github.com/mrmcwethy/Adafruit_CircuitPython_PCA9685.git

notthatbrad, 2017-09-25:
I'm pretty new to github, so not sure if I would be the best to do that, but I'll be happy to do anything I can. I can say that updating the motor.py file locally to 0x40 on my pc has corrected the error.

@mrmcwethy it failed looking for one on 0x40? Did you have a PCA9685 connected?

mrmcwethy, 2017-09-25:
@notthatbrad I don't have that board, so I **did** expect to get that error.  It just means i didn't make a big mistake and add syntax errors and i proved that the default address is now 0x40.  I am going to release my changes.

notthatbrad, 2017-09-25:
Groovy, that's why I was wondering. I had accidentally hooked mine up backwards earlier, and I got the error as well on code that I knew should be working.

tannewt, 2017-09-25:
Fixed in the driver commit: https://github.com/adafruit/Adafruit_CircuitPython_PCA9685/commit/c227c10f5424fb5d6859efb7aefcb72b4ff69305

I'll leave this open until we release a new version.

tannewt, 2017-12-27:
FYI, 0x60 makes sense for motors and stepper motors because its the default address of the motor featherwing. The servowing is 0x40 by default. I'm fixing this up with my rework of the PCA9685 drivers.

closed 2017-12-27 by tannewt

notthatbrad, 2017-12-28:
So it is (I just got the motor featherwing, and am testing it now)....but if I recall correctly, when using the PCA9685 breakout, the address is is 0x40, and the learn site has the featherwing documented as 0x40.
https://learn.adafruit.com/adafruit-8-channel-pwm-or-servo-featherwing/pinouts

This could be really confusing for people looking for the driver for the PCA9685 breakout

tannewt, 2017-12-28:
Yup, I'll be refining it shortly.