adafruit/Adafruit_CircuitPython_Bundle #33: error executing update-submodules

jerryneedell, 2017-11-05:
When I run ./update-submodules.sh I get an error in tls2561
Any suggestions?
```
jerryneedell@Ubuntu-Macmini:~/projects/adafruit_github/Adafruit_CircuitPython_Bundle$ ./update-submodules.sh 
Entering 'circuitpython'
Entering 'libraries/drivers/bme280'
Entering 'libraries/drivers/bme680'
Entering 'libraries/drivers/bmp280'
Entering 'libraries/drivers/bno055'
Entering 'libraries/drivers/ccs811'
Entering 'libraries/drivers/circuitplayground'
Entering 'libraries/drivers/dht'
Entering 'libraries/drivers/dotstar'
Entering 'libraries/drivers/ds3231'
Entering 'libraries/drivers/ht16k33'
Entering 'libraries/drivers/is31fl3731'
Entering 'libraries/drivers/lis3dh'
Entering 'libraries/drivers/max31855'
Entering 'libraries/drivers/max7219'
Entering 'libraries/drivers/mcp9808'
Entering 'libraries/drivers/mpr121'
Entering 'libraries/drivers/neopixel'
Entering 'libraries/drivers/pca9685'
Entering 'libraries/drivers/pcf8523'
Entering 'libraries/drivers/rgb-display'
Entering 'libraries/drivers/sdcard'
Entering 'libraries/drivers/si7021'
Entering 'libraries/drivers/ssd1306'
Entering 'libraries/drivers/thermistor'
Entering 'libraries/drivers/tsl2561'
Entering 'libraries/helpers/bus_device'
Entering 'libraries/helpers/hid'
Entering 'libraries/helpers/register'
Entering 'libraries/helpers/rtttl'
Entering 'libraries/helpers/simpleio'
Entering 'libraries/helpers/waveform'
Entering 'circuitpython'
Entering 'libraries/drivers/bme280'
Entering 'libraries/drivers/bme680'
Entering 'libraries/drivers/bmp280'
Entering 'libraries/drivers/bno055'
Entering 'libraries/drivers/ccs811'
Entering 'libraries/drivers/circuitplayground'
Entering 'libraries/drivers/dht'
Entering 'libraries/drivers/dotstar'
Entering 'libraries/drivers/ds3231'
Entering 'libraries/drivers/ht16k33'
Entering 'libraries/drivers/is31fl3731'
Entering 'libraries/drivers/lis3dh'
Entering 'libraries/drivers/max31855'
Entering 'libraries/drivers/max7219'
Entering 'libraries/drivers/mcp9808'
Entering 'libraries/drivers/mpr121'
Entering 'libraries/drivers/neopixel'
Entering 'libraries/drivers/pca9685'
Entering 'libraries/drivers/pcf8523'
Entering 'libraries/drivers/rgb-display'
Entering 'libraries/drivers/sdcard'
Entering 'libraries/drivers/si7021'
Entering 'libraries/drivers/ssd1306'
Entering 'libraries/drivers/thermistor'
Entering 'libraries/drivers/tsl2561'
usage: git rev-list [OPTION] <commit-id>... [ -- paths... ]
  limiting output:
    --max-count=<n>
    --max-age=<epoch>
    --min-age=<epoch>
    --sparse
    --no-merges
    --min-parents=<n>
    --no-min-parents
    --max-parents=<n>
    --no-max-parents
    --remove-empty
    --all
    --branches
    --tags
    --remotes
    --stdin
    --quiet
  ordering output:
    --topo-order
    --date-order
    --reverse
  formatting output:
    --parents
    --children
    --objects | --objects-edge
    --unpacked
    --header | --pretty
    --abbrev=<n> | --no-abbrev
    --abbrev-commit
    --left-right
    --count
  special purpose:
    --bisect
    --bisect-vars
    --bisect-all
Entering 'libraries/helpers/bus_device'
Entering 'libraries/helpers/hid'
Entering 'libraries/helpers/register'
Entering 'libraries/helpers/rtttl'
Entering 'libraries/helpers/simpleio'
Entering 'libraries/helpers/waveform'
```

dhalbert, 2017-11-06:
https://github.com/adafruit/Adafruit_CircuitPython_TSL2561.git does not have any releases at all, so there's no release that can be fetched.

jerryneedell, 2017-11-06:
hmm - so when I build - I see ```build-2.1.0/lib/adafruit_register 512
build-2.1.0/lib/adafruit_bus_device 512
build-2.1.0/lib/adafruit_hid 512
build-2.1.0/lib/adafruit_waveform 512
build-2.1.0/lib/adafruit_circuitplayground 512
build-2.1.0/lib/adafruit_pca9685 512
build-2.1.0/lib/adafruit_mpr121 512
build-2.1.0/lib/adafruit_lis3dh 512
build-2.1.0/lib/adafruit_ht16k33 512
build-2.1.0/lib/adafruit_rgb_display 512
build-2.1.0/lib/adafruit_ssd1306 512
build-2.1.0/lib/adafruit_max7219 512
fatal: No names found, cannot describe anything.
Stopping at 'libraries/drivers/tsl2561'; script returned non-zero status.
lib/README.txt 117 512
lib/update_linux.sh 646 1024
```

can I just ignore the errors?  tls2561.mpy is built

dhalbert, 2017-11-06:
The build script makes a list of all the libraries and their release tags. So that breaks when the script is run. The fix is just that we need to make a release for tsl2561. I'll do that in a minute.

dhalbert, 2017-11-06:
@jerryneedell After a ridiculous amount of to- and fro-ing, there's a release for you to try again with.

jerryneedell, 2017-11-06:
Thanks - it worked fine this time. Story for the trouble.

closed 2017-11-06 by jerryneedell