adafruit/Adafruit_CircuitPython_Bundle #68: Incorrect Link/Submodule

sommersoft, 2018-02-25:
Rolling through the doc updates, I discovered that the `drivers/ds18x20` module link points to `Adafruit_CirctuitPython_OneWire`. While a DS18x20 is a OneWire device, it does have it's own library...

MS Paint screenshot...
[image]

The hash listed on `drivers/ds18x20` matches the last commit to OneWire. 

3 days ago (from current writing), adabot updated OneWire twice using the same 3 PRs...not sure if that's the culprit, just the only thing I found out of the ordinary looking through the history.

ladyada, 2018-03-05:
oops same as

https://github.com/adafruit/Adafruit_CircuitPython_Bundle/commit/e208e059b9c9861acd36430d0ef0f39e9db2bbe5#commitcomment-27919947

tdicola, 2018-03-05:
Oh weird, yeah it must have been a copy paste mistake when adding those modules.  I just updated the .gitmodules to point at the right path.  If you already have the repository cloned you probably want to delete and re-clone with a full recursive clone to make sure you get the latest submodule state.  Apparently a 'git submodule sync' would also update the state but I tried it and it didn't do what I expected (very likely some odd submodule command incantation is required to make git happy--easier to just reclone).

closed 2018-03-05 by tdicola