adafruit/Adafruit_CircuitPython_Bundle #77: Remove Use of __init__.py Where Possible

sommersoft, 2018-05-24:
From discussion on Discord, it was decided we need to move away from having `__init__.py` files in every repo. This was previously required (I'm told), and is a CPython 2.x remnant.

There are non-empty `__init__.py` files in a few repos. Will need to investigate if they have code that needs/can be to moved, before deleting. They are:

- [ ] Thermal Printer

- [ ] RGB Display

- [ ] Character LCD

The other non-empty files only contain comments (license info, descriptions).

sommersoft, 2018-07-16:
@dhalbert, you found that `__init__.py` was actually still required for each repo, right? If so, you can close this...

dhalbert, 2018-07-16:
Nope, empty `__init__.py` is not be required. I went back and looked at this, because I couldn't remember the details. On June 4 I had trouble with `circuitpython-build-tools` because I had deleted `__init__`.py from https://github.com/adafruit/Adafruit_CircuitPython_CircuitPlayground and then it wouldn't build. But that turned out to be because the script was expecting >1 file in packages, and I fixed that.

https://github.com/adafruit/Adafruit_CircuitPython_CircuitPlayground and https://github.com/adafruit/Adafruit_CircuitPython_FancyLED don't have `__init__.py` files. And for a while the crickit library had multiple files but no `__init__.py`.

So, yes, it's safe to remember empty or comment-only `__init__.py` files.

kattni, 2018-12-29:
Subsequent library creation indicates that the results when dealing with packages is inconsistent without `__init__.py`. After discussion, it has been determined that we should include an `__init__.py` by default since that behavior is consistent. I'm closing this issue for now.

Further research is needed into why some packages function properly without the file and some do not. For now, inclusion is acceptable.

closed 2018-12-29 by kattni