adafruit/Adafruit_CircuitPython_Bundle #83: pip3* install circuitpython-build-tools

Coroecram, 2018-07-02:
The Readme.md specifies 

"pip install circuitpython-build-tools" but this returns: No matching distribution fund for circuitpython-build-tools. It should be, I believe:

"pip3 install circuitpython-build-tools"

tannewt, 2018-07-02:
@Coroecram Thanks for the suggestion! It depends on your setup. (It is pip if only python3 is installed.) Did you use the virtual environment?

Coroecram, 2018-07-03:
I was in the virtual environment step. In the end the tool did not work for me anyways and I could not run it even after using pip3.

I ended up having to copy the library I needed into my project, so I must have been doing something wrong.

I am running Raspbian (Stretch) with Python 2.7.13 and 3.5.3.

Perhaps I had to run pip update  with Python 2.7.13, but it is my understanding that the libraries were written for Python3, so it was odd that it wants pip for Python 2.

kattni, 2018-12-03:
@Coroecram Are you still experiencing issues? Please let us know so we can help you out or close this issue. Thanks!

Coroecram, 2018-12-03:
I did not get it to work. I ended up needing to import the libraries manually and was not able to get circuitpython to work following the documentation.

ladyada, 2018-12-03:
are you sure you wanted to install https://github.com/adafruit/circuitpython-build-tools ? that's our internal tool for development/management. individual libraries can be installed with pip3

closed 2018-12-28 by ladyada