adafruit/Adafruit_CircuitPython_Bundle #94: Add mpy version number to lib folder name

kattni, 2018-09-30:
From @dhalbert: ...Include the mpy version in the name, like lib2, lib3, etc. and customize sys.path accordingly in the build. That helps solves the major-version update problem.

tagging:
@tannewt
@ladyada

dhalbert, 2018-09-30:
It's an historical accident that the mpy version (2,3) coincides with the 2.x and 3.x releases.

ladyada, 2018-09-30:
ok im not 100% sure what we're solving anymore. lets chat monday!

tannewt, 2018-10-01:
What is the context for this?

dhalbert, 2018-10-01:
The point of this is that when the .mpy version changes, the old `lib` directory becomes unusable. I've seen a number of support cases where it took some time for someone upgrading from CircuitPython 2.x to 3.x not realizing they had to update `lib`. Sometimes their `main.py` crashed hard enough that the board was unusable and they had to reinstall 2.x temporarily.

If the lib directories included the mpy version (`lib2`, `lib3`), and only that directory was on `sys.path`, then a mismatched `lib` directory wouldn't be accessed at all. There'd still be an error, but it might be less confusing. This is vaguely similar to how there are separate library directories for CPython 2.x and 3.x on host OS's.

One issue is that we're not going to see an mpy version bump in CPy 4.0, so the mpy version will no longer (accidentally) line up with the CPy version. The digit in `lib2` and `lib3` could refer to the mpy version or the CPy version. Since there's no mpy version bump from CPy 3.x to 4.x (probably), maybe the lib directory should be called `lib34`. On the other hand, there may be other differences between libraries between 3.x and 4.x (e.g. changing `UART` timeout from msecs to seconds), and maybe we want to think about library versions in the long run.

I floated this more as an idea to spur thinking about library version management -- I'm not sure if this particular idea is the best, but it solves a particular support problem.

kattni, 2018-10-01:
After discussion we've decided the best route to go is to add a specific flash code into the CircuitPython error flash codes. We're going to add pulsing blue to be specific to the Incompatible mpy error to resolve this.

kattni, 2018-12-03:
Closing in favor of: https://github.com/adafruit/circuitpython/issues/1369

closed 2018-12-03 by kattni