adafruit/Adafruit_CircuitPython_Bundle #100: Consistent naming of archive files?

mew-cx, 2018-10-17:
Hi,

I realize this is a nit, but is there a reason why the source archives are named as camelcase and underscores, while the mpy archives are named as lowercase and hyphens?  Also, the date appears in different locations between those archive types.

On a case-sensitive OS like Linux, I wind up renaming the camelcase so the files appear together in a list.

A minor thing, I know :-)
cheers
-- mew

dhalbert, 2018-10-17:
I'm not sure what you mean. Could you point to the files that are named inconsistently. The bundle release assets have consistent naming: https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/latest.

mew-cx, 2018-10-17:
Sure.  These are the filenames I get when downloading w Chrome (dont know if that matters).  The first two are camelcase and underscore, the following three are lowercase and hyphen.  If they use the same capitalization and punct, with the date after "bundle", they would sort together with locale LC_ALL=C.
cheers
-- mew

```
mew@spin:circuitpython !4 0 $ ll
total 2404
-rw-rw-r-- 1 mew mew    8648 Oct 16 20:26 Adafruit_CircuitPython_Bundle-20181016.tar.gz
-rw-rw-r-- 1 mew mew   33414 Oct 16 20:27 Adafruit_CircuitPython_Bundle-20181016.zip
-rw-rw-r-- 1 mew mew  525579 Oct 16 20:26 adafruit-circuitpython-bundle-2.x-mpy-20181016.zip
-rw-rw-r-- 1 mew mew  525579 Oct 16 20:26 adafruit-circuitpython-bundle-3.x-mpy-20181016.zip
-rw-rw-r-- 1 mew mew 1315076 Oct 16 20:26 adafruit-circuitpython-bundle-py-20181016.zip
mew@spin:circuitpython !5 0 $ 
```

dhalbert, 2018-10-17:
Ah, I see. The capitalized files are not named by us, but are generated automatically by GitHub when we make a release (highlighted below). There's no way to turn those off. We don't actually recommend that you download those files to get the source because they contain stuff that you don't want or need in your lib directory on CIRCUITPY. The "bundle-py" contains the .py versions that correspond to the .mpy versions.

[image]

mew-cx, 2018-10-17:
Ah, ok.  Thanks.

closed 2018-10-17 by mew-cx