adafruit/Adafruit_CircuitPython_Bundle #121: Installation issues on Raspberry Pi/Raspbian

easleyjs, 2018-12-30:
Hey guys,

I'm trying to install the bundle on a Raspberry Pi 3 B to use the max7219 driver in a CircuitPy project. After installing Raspbian, and following the Adafruit tutorial to install CircuitPython, I tried installing the bundle using the recommended pip command:

**pip3 install adafruit-circuitpython-lis3dh**

It appeared to download the necessary dependencies, and install correctly, but on inspection of /usr/local/lib/python3.5/dist-packages - I only see Blinka, CoreIO, and Adafruit GPIO. When I try to import the max7219 module in Mu, I get the following error:

ImportError                               Traceback (most recent call last)
<ipython-input-1-18ce24a97021> in <module>()
----> 1 import max7219

ImportError: No module named 'max7219'

uname -a on my RPi:
Linux raspi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

Python version info from Mu:
Jupyter QtConsole 4.2.1
Python 3.5.3 (default, Sep 27 2018, 17:25:39)

Thanks for any help you can provide.

ladyada, 2018-12-30:
`pip3 install adafruit-circuitpython-lis3dh` is the `lis3dh` driver, not the max7219?

ladyada, 2018-12-30:
yes, see the "For example" text before then :)
if you're having an issue with the MAX7219 you can post an issue in https://github.com/adafruit/Adafruit_CircuitPython_MAX7219 
will close this for now since its not a bundle issue!

closed 2018-12-30 by ladyada