adafruit/Adafruit_CircuitPython_Bundle #124: PulseIO problem on Raspberry Pi when running irremote_simpletest.py examplr

macd2point0, 2019-01-22:
I installed the entire CircuitPython on my Raspberry Pi running Raspian Stretch in the /lib/ directory. When trying to run the irremote_simpletest.py example, I encounter the following:

```
Traceback (most recent call last):
  File "irremote_simpletest.py", line 3, in <module>
    import pulseio
  File "/home/pi/.local/lib/python3.5/site-packages/pulseio.py", line 4, in <module>
    from adafruit_blinka.microcontroller.bcm283x.pulseio.PulseIn import PulseIn
ImportError: No module named 'adafruit_blinka.microcontroller.bcm283x.pulseio'
```

This was a brand new installation of Raspian, and I precisely followed the steps in https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

brennen, 2019-01-22:
Yeah, I think this should be good to go with a fresh install, but I will attempt to reproduce.

brennen, 2019-01-22:
So as an update here:

- something seems a bit off with the blinka install; I thought maybe I'd just mis-tagged the release, but that doesn't seem to be it. Investigating further.
- Adafruit_CircuitPython_IRRemote is missing a couple of dependencies, will put in a PR for that one.

macd2point0, 2019-01-23:
At least it wasn't user error...yet. Thanks for the update.

dhalbert, 2019-10-28:
Closing as stale. Let us know if there are still packaging issues.

closed 2019-10-28 by dhalbert

espenseventyr, 2019-11-13:
Hello. I still get "AttributeError: module 'pulseio' has no attribute 'PWMOut' when I try to use the pulseio.PWMOut function. I'm using a Raspberry Pi 3. Is PWM still not supported on RPI, or is there something I can do to make it work?

ladyada, 2019-11-13:
it does not, and it is not trivial to add