adafruit/Adafruit_CircuitPython_Bundle #128: Incompatible mpy files

dakhunter, 2019-01-30:
I'm trying to install the Adafruit library for ServoKit on my ESP8266 nodeMCU. I have tried to install on both CircuitPython and MicroPython per this tutorial
`https://learn.adafruit.com/micropython-hardware-pca9685-pwm-and-servo-driver?view=all`
However, I kept running to the Incompatible .mpy files for both of them. On CircuitPython, I'm using Adafruit CircuitPython 3.1.2 and the CircuitPython_Bundle-3.x on the Release of the GitHub page
Here's the returned error when I import the file
[image]
And for MicroPython, I use esp8266-20180511-v1.9.4 version. I tried And I ran into the same issue as well. 
Any help would be great :D

dhalbert, 2019-01-30:
Make sure you don't have another copy of `adafruit_pca9685` in the top-level of CIRCUITPY - it will choose that one first. Also, when you copy the bundle, make sure you do an Eject on Windows or a `sync` on Linux to ensure that all the files have been completely copied before resetting the board or unplugging it.

dakhunter, 2019-01-30:
I only have one copy of the adafruit_pca9685 at a time. I'm using ESPlorer to send files over my ESP8266. The files I think are at the top level with the boot.py and main.py. 
When I send the code to the ESP, the tool allows me to import the file. But when I save the file to the ESP, I cannot import the module.
Not sure where the error comes from here

dakhunter, 2019-02-02:
So apparently transferring with ESPlorer mess up my file format somehow. When I transfer with webrepl, it let me import the modules for servos.
And it only works with micropython 1.8.x sadly

ladyada, 2019-02-02:
our libraries are not micropython compatible - circuitpython only :)

ladyada, 2019-02-19:
@dakhunter hiya is this resolved? we can close it?

dakhunter, 2019-02-19:
Yes. It is resolved. I stuck with the older version of microPython and older version of Adafruit library and they worked like charm 💃

closed 2019-02-19 by ladyada