adafruit/Adafruit_CircuitPython_Bundle #156: PulseIO pwm output problem on Raspberry  pi 3

cetinkayafaruk, 2019-04-15:
I tried to run pwm example circuit python on raspberry pi but I am getting some error. How can I solve the problem ? Is Raspberry pi supported PulseIO library?

**Example :** 
https://circuitpython.readthedocs.io/en/3.x/shared-bindings/pulseio/PWMOut.html

import pulseio
import board

pwm = pulseio.PWMOut(board.D16, frequency=50)
pwm.duty_cycle = 2 ** 15

**Output:** 
  Traceback (most recent call last):
  File "/home/pi/Desktop/circuitpython_test_pwm.py", line 5, in <module>
    pwm = pulseio.PWMOut(board.D16)     # output on D16
AttributeError: module 'pulseio' has no attribute 'PWMOut'

deshipu, 2019-04-15:
Do you have a file called ``pulseio.py`` somewhere on your device?

ladyada, 2019-04-15:
pulseio isnt suported on Pi yet

ladyada, 2019-04-15:
https://github.com/adafruit/Adafruit_CircuitPython_Bundle/issues/124 dup

closed 2019-04-15 by ladyada