adafruit/Adafruit_CircuitPython_Bundle #162: Essentials bundle?

tannewt, 2019-05-22:
As the bundle grows past 2MB should we provide an essentials bundle that is just core deps like Bus Device? Maybe each library zip should include dependencies instead?

brentru, 2019-05-22:
I like the idea of library zips including dependencies since they're already accounted for in `requirements.txt`

caternuson, 2019-07-10:
I'll 2nd the library zip idea using requirements.txt.

Even better would be some kind of mechanism that would allow getting all the libraries needed for a specific guide. Something like a requirements.txt that the guide author would have to create, but then there would be a single click download option available in the guide.

dhalbert, 2020-07-05:
This is related to https://github.com/adafruit/circuitpython-org/issues/491