adafruit/Adafruit_CircuitPython_Bundle #171: SD card module error with Circuit Python 4.0.2

AidanTek, 2019-07-01:
Hello, I am trying to use a micro sd card with a breakout and the Feather M0 Express.

I have verified I am running Circuit Python 4.0.2, and I have the bundle-4.x 

I get this error when I try to import the module:

```
>>> import adafruit_sdcard
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_sdcard.py", line 60, in <module>
MpyError: Incompatible .mpy file. Please update all .mpy files. See http://adafru.it/mpy-update for more info.
>>> 
```

jerryneedell, 2019-07-01:
You may just have a problem reading the file. Try reloading the .mpy files to the boards and make sure the board has finished all writes to the board before attempting to run any code.
If you are on Linux - do a “sync”. For other OS, I’m not certain of the best ways, but you can eject then remount CIRCUITPY or just wait at least a minute....

AidanTek, 2019-07-01:
Aha, yes waiting for 60 seconds seems to do the trick, thanks @jerryneedell

AidanTek, 2019-07-01:
Closing was premature sorry, a second import looked ok, but it just fails to produce the error.

Using the example:

```
import os
import busio
import digitalio
import board
import storage
import adafruit_sdcard

# The SD_CS pin is the chip select line.
#
#     The Adalogger Featherwing with ESP8266 Feather, the SD CS pin is on board.D15
#     The Adalogger Featherwing with Atmel M0 Feather, it's on board.D10
#     The Adafruit Feather M0 Adalogger use board.SD_CS
#     For the breakout boards use any pin that is not taken by SPI

SD_CS = board.D10 # setup for M0 Adalogger; change as needed

# Connect to the card and mount the filesystem.
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
cs = digitalio.DigitalInOut(SD_CS)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

# Use the filesystem as normal! Our files are under /sd

# This helper function will print the contents of the SD
def print_directory(path, tabs=0):
    for file in os.listdir(path):
        stats = os.stat(path+"/"+file)
        filesize = stats[6]
        isdir = stats[0] & 0x4000

        if filesize < 1000:
            sizestr = str(filesize) + " bytes"
        elif filesize < 1000000:
            sizestr = "%0.1f KB" % (filesize/1000)
        else:
            sizestr = "%0.1f MB" % (filesize/1000000)

        prettyprintname = ""
        for _ in range(tabs):
            prettyprintname += "   "
        prettyprintname += file
        if isdir:
            prettyprintname += "/"
        print('{0:<40} Size: {1:>10}'.format(prettyprintname, sizestr))

        # recursively print directory contents
        if isdir:
            print_directory(path+"/"+file, tabs+1)

print("Files on filesystem:")
print("====================")
print_directory("/sd")

```

The program breaks when the module is accessed:

```
sdcard = adafruit_sdcard.SDCard(spi, cs)
```

AidanTek, 2019-07-01:
Hello again, I found the problem - I was using the adafruit_bus_device module that came included with 4.0.2 - it seems there is an incompatibility between the 4.0.2 adafruit_bus_device module and the bundle 4x adafruit_sdcard module.

Fixed by replacing with the adafruit_bus_device module from the bundle

closed 2019-07-01 by AidanTek