adafruit/Adafruit_CircuitPython_Bundle #184: Is that feasible? Compatible micropython

Pillar1989, 2019-08-21:
I tried to modify the link here as micropython url. and rebuild the Adafruit_CircuitPython_Bundle. Some hardware independent libraries actually work on pyboard, such as binascii ...

https://github.com/adafruit/circuitpython-build-tools/blob/519c62f3aa868e02aff9f62ffbe85e4ade6825ac/circuitpython_build_tools/build.py#L70-L74

Are there plans to supplement these libraries for the micropython software ecosystem in the future?

ladyada, 2019-08-21:
yes for hardware support, we have an interface shim layer here
https://github.com/adafruit/Adafruit_Blinka
it (at least used to) run on pyboard, but we haven't tried it in a bit :)

tannewt, 2019-08-22:
The mpy files created by the script should work in their corresponding MicroPython version. There is no need to change the script to MicroPython.

However, the hardware APIs used by the libraries themselves are different between the two. Blinka provides the CircuitPython APIs in CPython on Linux and on MicroPython. It's not used often on MicroPython so it may not work.

What platform are you using MicroPython on? Could you run CircuitPython instead?

dhalbert, 2019-10-28:
Closing for now. Happy to reopen if you have further questions.

closed 2019-10-28 by dhalbert