adafruit/Adafruit_CircuitPython_Bundle #191: Neotrellis library install issue on Raspberry Pi

dpjenkin, 2019-09-02:
@ladyada I'm late to the party here, but I have a similar issue that is driving me crazy. I'm trying to install the Neotrellis library using:

_sudo pip3 install adafruit-circuitpython-neotrellis_

It kicks back with an error like @sUpniverse posted above. Using the tutorial here (https://learn.adafruit.com/adafruit-neotrellis?view=all) it tells me to try using "pip" instead if "pip3" didn't work. I tried "pip" and got an error like "Cannot find a version that satisfies the requirement Adafruit-Blinka"

Any thoughts? I've updated the Pi and rebooted already. I've installed Python 3 and Blinka as well.

_Originally posted by @dpjenkin in https://github.com/adafruit/Adafruit_CircuitPython_Bundle/issues/117#issuecomment-527136220_

ladyada, 2019-09-02:
hiya please post the complete output and any other info. you should be using python3 for sure, please use `pip3` and/or verify you're using python3!

dpjenkin, 2019-09-02:
I figured it out...wasn't running Python 3. I finally see the light! Thanks for the help

closed 2019-09-02 by ladyada