adafruit/Adafruit_CircuitPython_Bundle #195: Update drivers, helpers, examples using wait_for_frame, refresh_soon

anecdata, 2019-09-06:
grep of Library Bundle shows those two calls (wait_for_frame, refresh_soon) used in:
```
pyportal (drivers and helpers)
miniqr/examples/
turtle (drivers and helpers)
display-text/examples/ (drivers and helpers)
pybadger
pyoa (drivers and helpers)
slideshow
```

(Learn guides also extensively use the pre-5.x refresh paradigm)

jerryneedell, 2019-09-06:
slideshow has been updated - new release yesterday...

makermelissa, 2019-11-13:
It appears the proper replacement for `wait_for_frame()` is `refresh(target_frames_per_second=60)`
To make compatible with both CP 4.1 and CP 5.0, it looks like the following code block works:
```
try:
    display.refresh(target_frames_per_second=60)
except AttributeError:
    display.wait_for_frame()
```

makermelissa, 2019-11-27:
It should now all be updated. For a lot of the code, a simple grep may still show wait_for_frame() or refresh_soon(), which are in try blocks so that the code also works in 4.0.

closed 2019-11-27 by makermelissa