adafruit/Adafruit_CircuitPython_Bundle #221: Trellis M4 is broken buf is too small. need 1056 bytes

Javantea, 2020-01-18:
Running the Launch Deck code from https://learn.adafruit.com/launch-deck-trellis-m4/code-with-circuitpython I get this error:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_trellism4.py", line 250, in __init__
  File "adafruit_trellism4.py", line 57, in __init__
  File "neopixel.py", line 117, in __init__
ValueError: buf is too small. need 1056 bytes
```

Reproducing this error line by line it occurs on:
```python
trellis = adafruit_trellism4.TrellisM4Express(rotation=ROTATION)
```
To get to a version that works, I had to go back to Aug 28, 2019 version and fix this bug:
```
Traceback (most recent call last):
  File "code.py", line 84, in <module>
TypeError: function takes 1 positional arguments but 2 were given
```
which then fixes the issue.
The fix to the Aug 28, 2019 issue is to replace the keyboard and consumer control with:
```python
kbd = Keyboard()
cc = ConsumerControl()
```
but that is tangential so that perhaps other users can get their Trellis M4 to work.

ladyada, 2020-01-18:
@Javantea do other examples work? we'll take a look - thanks :)

Javantea, 2020-01-18:
I just tested Eight-Step Drummer with January 17, 2020 auto-release and got the same error:
```
code.py output:
Traceback (most recent call last):
  File "code.py", line 37, in <module>
  File "adafruit_trellism4.py", line 250, in __init__
  File "adafruit_trellism4.py", line 57, in __init__
  File "neopixel.py", line 117, in __init__
ValueError: buf is too small. need 1056 bytes
```
I suspect that anything that calls `trellis = adafruit_trellism4.TrellisM4Express(rotation=ROTATION)` is affected

ladyada, 2020-01-18:
hiya so we can repro- what's the output of boot_output.txt?

Javantea, 2020-01-18:
```
Adafruit CircuitPython 4.1.2 on 2019-12-18; Adafruit Trellis M4 Express with samd51g19
```

ladyada, 2020-01-18:
can you try 5.x beta? :)

Javantea, 2020-01-18:
It works, thanks!

ladyada, 2020-01-18:
yay!

closed 2020-01-18 by ladyada