adafruit/Adafruit_CircuitPython_Bundle #250: I2CDevice incompatible

fvzeppelin, 2020-05-28:
Setting:
- Device: ItsyBitsy M0 Express
- CircuitPython 5.3.0
- adafruit-circuitpython-bundle-5.x-mpy-20200528

This line un mu:
`from adafruit_bus_device.i2c_device import I2CDevice`
causes:
`MpyError: Incompatible .mpy file. Please update all .mpy files. See http://adafru.it/mpy-update for more info.`
when saving to code.py

Expected: COmpiling without error.

caternuson, 2020-05-28:
How did you update the firmware to 5.3.0 and install the libraries from the bundle file?

Just tested and everything seems OK with all the latest:
```
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit ItsyBitsy M0 Express with samd21g18
>>> from adafruit_bus_device.i2c_device import I2CDevice
>>> 
```
Can you access the REPL via the serial console and try that same test and post what you get.

fvzeppelin, 2020-05-28:
I have installed the latest bootloader as described here:
[](https://learn.adafruit.com/introducing-itsy-bitsy-m0/uf2-bootloader-details)
And updated Circuitpython from here:
[](https://circuitpython.org/board/itsybitsy_m0_express/)
Ouput:
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit ItsyBitsy M0 Express with samd21g18
>>> from adafruit_bus_device.i2c_device import I2CDevice
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MpyError: Incompatible .mpy file. Please update all .mpy files. See http://adafru.it/mpy-update for more info.
>>> 
```

caternuson, 2020-05-28:
What about the library files? After downloading `adafruit-circuitpython-bundle-5.x-mpy-20200528.zip`, what did you do?

fvzeppelin, 2020-05-28:
Ah, as you say it, I have noticed:
I was so happy that `adafruit_bus_device` was already installed that I did not update it - call me a noob!
Now that I've updated it, everything works :)
Thank you for your super-fast responses!

closed 2020-05-28 by fvzeppelin

caternuson, 2020-05-28:
Yeah! Glad it was something simple and easy. Have fun!