adafruit/Adafruit_CircuitPython_Bundle #262: Add library documentation links

woolseyj, 2020-09-16:
Would it be possible to add the readthedocs documentation links for all the libraries to the circuitpython_library_list.md file?  It would be nice to have all of a library's associated links (GitHub, Documentation, PyPi, Etc.) in one place.

In addition, or instead of, it would also be very nice to add the readthedocs documentation link to the top level README for each library's GitHub repository.

tannewt, 2020-09-16:
It should be relatively easy to add all doc links to that page because we mostly use a standard format.

I'm not sure what you mean by your second point. Could you provide an example line you'd like to see in the file?

woolseyj, 2020-09-16:
Great!

For the second part, let me provide an example for clarification.  Say you go to the individual GitHub repository for the NeoPixel library (https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel).  At the bottom of the top level README under the Documentation section, I would expect to see a link to the API documentation (https://circuitpython.readthedocs.io/projects/neopixel/) for that library.  If not in the Documentation section, at least somewhere on that page.  Does this make sense?

woolseyj, 2020-09-16:
Oops, never mind that second part.  I missed the fact that it is already included in the docs/passing button/tag.  Sorry for the confusion.

tannewt, 2020-09-17:
@woolseyj are you interested in making a PR to add the doc links to this file? Once we get caught up, we can make sure new items have it when added.

woolseyj, 2020-09-17:
Sure, but it will probably have to wait a few days.

woolseyj, 2020-09-18:
Just created the pull request.

woolseyj, 2020-09-18:
By the way, a couple of the repositories had links to documentation, but it does not appear the documentation had yet been created.

https://github.com/adafruit/Adafruit_CircuitPython_BluefruitConnect
https://github.com/adafruit/Adafruit_CircuitPython_IterTools

tannewt, 2020-09-21:
Fixed by #265. Thanks, I'll take a look at those repos.

tannewt, 2020-09-22:
I made a PR for itertools and fixed the RTD setup for bluefruit connect. Thanks!

woolseyj, 2020-11-06:
It does not appear this was ever resolved and needs to be reopened.  See [PR](https://github.com/adafruit/Adafruit_CircuitPython_Bundle/pull/265).

FoamyGuy, 2020-11-06:
I just recently made some other tweaks to some of the Adabot scripts so I'm all set up to do that now. I can try to make the changes necessary in adabot to get this page to build with the docs links in it.

woolseyj, 2020-11-06:
Thank you for looking into this.  You can move this issue to another repo if you want, I just want to make sure we don't lose the original information in this issue or the associated PR.

FoamyGuy, 2020-11-13:
Fixed via adabot PR above. The md file now gets generated with docs links in it.

closed 2020-11-13 by FoamyGuy

woolseyj, 2020-11-13:
Excellent!  Thank you so much.