adafruit/Adafruit_CircuitPython_Bundle #270: Feature Request: BLE HID

street-grease-coder, 2020-10-20:
BLE mouse/keyboard would be a nice addition!

tannewt, 2020-10-20:
We support it in the BLE library plus the HID library. See this example: https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/master/examples/ble_hid_periph.py

closed 2020-10-20 by tannewt

street-grease-coder, 2020-10-21:
I didn't find an example for a BLE mouse, that would be very helpful

FoamyGuy, 2020-10-21:
@street-grease-coder there is a mouse example here: https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/master/examples/hid_simpletest.py you'd have to do a bit of mashing together. But if you take the BLE setup from the example linked above and combine it with the hid mouse code in this example I linked I think it should work as a basic BLE HID mouse example. 

If you get that working maybe you can make a PR on one of these libraries to add it as an extra example.