adafruit/Adafruit_CircuitPython_Bundle #275: digitalio not found in Python library

Fiftyohms, 2020-10-31:
import digitalio  Python -Thonny emvironment

results-
Traceback (most recent call last):
  File "/home/pi/Desktop/AM2303DHT.py", line 31, in <module>
    import digitalio
ModuleNotFoundError: No module named 'digitalio'

What am I not understanding..
Thanks,
Jeff

FoamyGuy, 2020-10-31:
You need to install Blinka the Circuit Python compatibility layer. There is a guide that covers the process here: https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

Once you get that all set up your python should have access to `digitalio` provided by Blinka.

siddacious, 2020-10-31:
@Fiftyohms It looks like you are trying to use the AM2303 with a raspberry pi and the Thonny editor?

 @FoamyGuy has the right idea, however I don't believe the CircuitPython DHT library supports the AM2303; furthermore, I believe the DHT/AM230x sensors do not work well on the raspberry pi in general so i wouldn't be surprised if you have issues getting it to work.

Once you have the prerequisites for Blinka installed, you can try installing the CircuitPython DHT library via PyPi, using the instructions here:
https://pypi.org/project/adafruit-circuitpython-dht/

**again, I suspect it will not work as expected** with the AM2303; you may wish to try a different humidity sensor, such as one of these offerings from Adafruit:

https://www.adafruit.com/?q=humidity+stemma&sort=BestMatch

Lastly, I would suggest using the Mu editor when working with circuitpython. Like Thonny, it's targeted towards beginners and it comes with CircuitPython support built in:
https://codewith.mu/en/download

Good luck!

closed 2020-10-31 by siddacious

Fiftyohms, 2020-11-01:
I still get the same error when I run the AdafruitDHT.py from my project directory.
Cannot find Adafruit_DHT...
Followed every step to install BLINKA package.
Jeff

siddacious, 2020-11-02:
@Fiftyohms github is not meant to be used for technical support; please make a post here describing what you are trying and what the exact errors you are getting in as much detail as possible, including the exact code you are running. Additionally well lit clear and in-focus pictures  of your wiring that show all the connections will be needed to rule out any hardware issues.

https://forums.adafruit.com/viewforum.php?f=60