adafruit/Adafruit_CircuitPython_Bundle #311: Verify that Gyro/Accel Libraries are returning the correct units (SI)

jposada202020, 2021-05-08:
Was wound in https://github.com/adafruit/Adafruit_CircuitPython_L3GD20/issues/20 returning value units were not as expected. This Issue will keep track of the progress in reviewing other libraries and to register findings and corrections

jposada202020, 2021-05-08:
Checks
------------

| Name   |      Library      |  Datasheet | Units |
|----------|:-------------:|:-----:|:---:|
| BNO055|  [Adafruit BNO055](https://github.com/adafruit/Adafruit_CircuitPython_BNO055)| [See](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bno055-ds000.pdf)|rad/s |
| BN088 |    [Adafruit BNO08X](https://github.com/adafruit/Adafruit_CircuitPython_BNO08x)  |  [See](https://www.ceva-dsp.com/wp-content/uploads/2019/10/BNO080_085-Datasheet.pdf)|rad/s |
|LSM9DS1|[Adafruit LSM9DS1](https://github.com/adafruit/Adafruit_CircuitPython_LSM9DS1)|[See](https://www.st.com/resource/en/datasheet/lsm9ds1.pdf)|deg/s
|LSM6DSOX|[Adafruit LSM9DS1](https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS)|[See](https://www.st.com/resource/en/datasheet/lsm6dsox.pdf)|rad/s|
|L3GD20H|[Adafruit L3GD20H](https://github.com/adafruit/Adafruit_CircuitPython_L3GD20)|[See](https://www.st.com/en/mems-and-sensors/l3gd20.html)|deg/s|

To verify
---------

* Adafruit TDK InvenSense ICM-20948 9-DoF IMU (MPU-9250 Upgrade) - STEMMA QT / Qwiic
* Adafruit MPU-6050 6-DoF Accel and Gyro Sensor - STEMMA QT Qwiic
* Adafruit Precision NXP 9-DOF Breakout Board - FXOS8700 + FXAS21002
* Adafruit ICM-20649 Wide Range ±30g ±4000dps 6-DoF IMU - STEMMA QT / Qwiic

ladyada, 2021-05-08:
oi vey - yeah they will need to be updated :(

jposada202020, 2021-05-09:
I am on it, already did the L3GD20H and will apply same principle to the other two.

jposada202020, 2021-05-09:
All changes were made. See 
https://github.com/adafruit/Adafruit_CircuitPython_L3GD20/pull/23
https://github.com/adafruit/Adafruit_CircuitPython_MPU6050/pull/17
https://github.com/adafruit/Adafruit_CircuitPython_LSM9DS1/pull/30

closed 2021-05-09 by jposada202020