adafruit/Adafruit_CircuitPython_Bundle #326: The bundle does not fit on a NUCLEO-H743ZI2

mickpf, 2021-07-08:
Hello,

It's my second (the first one was a couple of months ago) and probably my last try to reach a goal with (Ciruit-)Python.
I've installed the latest CircuitPython 6.3.0 and try to place the lib directory of the ZIP file "adafruit-circuitpython-bundle-6.x-mpy-20210707.zip" in the pseudo-device "CURCUITPY". My host (Ubuntu 20.04) says: Not enough space (372,7 kB available, 2.4 MB is required).

I hate Python...

Do I something wrong or is CircuitPython not really tested with this board?

Thanks in advance,
Michael

dhalbert, 2021-07-08:
The bundle has outgrown flash. The whole bundle is not meant to be placed on a board any more. This has been true for a few years. There are almost no boards it will fit on. https://learn.adafruit.com/welcome-to-circuitpython/circuitpython-libraries#copying-libraries-to-your-board-2978253-18 has more details.

dhalbert, 2021-07-08:
If you need more help, please ask in https://adafru.it/discord or the forums: https://forums.adafruit.com. Thanks.

closed 2021-07-08 by dhalbert

mickpf, 2021-07-08:
I'd like to connect (and test) an Adafruit AirLift (for WiFi) with the NUCLEO baord. I am familiar with ESP32 and ESP-IDF.
Could you tell me please, which files and subdirectories I need for this goal.

Thanks in advance,
Michael

dhalbert, 2021-07-08:
Please do not ask support questions here. Please ask in the support locations I gave.

mickpf, 2021-07-08:
Could you give me the URL please?

mickpf, 2021-07-08:
Sorry, I didn't see it above...

dhalbert, 2021-07-08:
> If you need more help, please ask in https://adafru.it/discord or the forums: https://forums.adafruit.com. Thanks.

as I said above