adafruit/Adafruit_CircuitPython_Bundle #327: ImportError: no module named 'gamepad'

lukasparsons, 2021-07-10:
I may be doing something wrong here because I am new with circuitpython and this type of development in general. I come from a C# background.

I am using the latest version 7.x of circuitpython and I am getting an error that the gamepad module cannot be found. I looked into adafruit_hid and it seems that gamepad was removed, so I attempted get an older version of adafruit_hid and include gamepad.mpy, but that didn't work. Lastly I tried a custom build where I removed adafruit_hid from the build and I am still getting this error.

Like I said, I may be missing something due to being new to CircuitPython, but if I had to guess where my error was I would think it has something to do with not properly removing it from the custom build.

Thank you in advance for your help.

dhalbert, 2021-07-10:
Built-in gamepad support has been removed from CircuitPython 7, because it is not well supported on several host operating systems. It is possible to define your own HID devices in 7. You can still use the old gamepad with CircuitPython 6.3.0 and an older version of the library: https://github.com/adafruit/Adafruit_CircuitPython_HID/releases/tag/4.3.0

lukasparsons, 2021-07-10:
I should have been more descriptive.  I am not using gamepad anywhere in my code and I am receiving this error. the build is a custom build of the latest version of  circuitpython at the time of posting, 7a4.

lukasparsons, 2021-07-10:
I'm going to close this as I found that I believe I needed to check out using the 7.0 release tag. I'm going to see if I can do a build now while enabling usb_cdc. If I have any questions I will open them on the circuitpython repo as I don't think this is the place for that. Thank you.

lukasparsons, 2021-07-10:
gamepad seems to still be referenced in circuit_playground_base.mpy at 7.x-mpy-20210709\lib\adafruit_circuitplayground

I downloaded the zip with .py files to be sure. It is referenced on line 34 of circuit_playground_base.py as I said before I may be missing something or ignorant of something, or it may just be that circuitplayground libraries arent updated to be compatible with 7.x yet.

dhalbert, 2021-07-10:
If your board is using frozen modules, then the problem may be that you have not updated the submodules in your repo clone. You may have an old HID commit in `frozen/`. Do `git submodule update --init`.

lukasparsons, 2021-07-10:
That very well may have been my issue. I have resolved it for now by downgrading CP to 6.3.x and enabling usb_cdc in that build. Seems to have resolved the issue. I was just trying to get the usb_cdc bug fixes in the latest build, but I have it running now how I intended, and my code is running as expected.

Thanks for your help.

closed 2021-07-10 by lukasparsons

e-simpson, 2021-07-23:
I'm also having issue with "gamepad" references on an Adafruit Clue board (nRF52) as I try to get the 7.0.0 alpha running.