adafruit/Adafruit_CircuitPython_Bundle #329: PyPI variants of the library have `__version__ = "0.0.0-auto.0"`

aivarannamaa, 2021-07-14:
It looks like the step of assigning the actual version is skipped for the modules uploaded to PyPI. Problem occurs at least with https://pypi.org/project/adafruit-circuitpython-framebuf/#files but I believe I've seen this with other libraries as well.

lesamouraipourpre, 2021-08-09:
This has been fixed for newly created libraries (when created using `cookiecutter`) and there is a pending PR (https://github.com/adafruit/cookiecutter-adafruit-circuitpython/pull/146) to improve it.

So far, there has been no push to go through the libraries and make the changes to existing libraries. This will hopefully happen after that PR is merged and an `adabot` patch script can be created to make the changes to all the libraries at once.

jepler, 2022-06-08:
Just want to bump this as something we might want to roll out to all libs now that it's been in cookiecutter for awhile @kattni

tekktrik, 2022-08-25:
Now patched!  PyPI variants now have the correct tag version!

closed 2022-08-25 by tekktrik

dhalbert, 2022-08-25:
Yay!