adafruit/Adafruit_CircuitPython_Bundle #336: Should more libraries be pushed to PyPI for Blinka?

lesamouraipourpre, 2021-08-09:
Below is a list of bundle libraries containing a `setup.py.disabled`

I've separated them into 3 groups:

1. Libraries which cannot be used by Blinka and shouldn't be pushed to PyPI. These appear to all be hardware specific, eg. the MacroPad.
2. Libraries which can be used by Blinka and should be pushed to PyPI. These initially appear to be drivers for hardware that could be connected to a Blinka device.
3. Libraries that I am unsure if Blinka could make use of. I intend to investigate these further and re-categorize them until this group is empty.

## Don't push to PyPI - Blinka can't use it
* drivers/circuitplayground/setup.py.disabled
* drivers/clue/setup.py.disabled
* drivers/monsterm4sk/setup.py.disabled
* drivers/pyportal/setup.py.disabled
* helpers/funhouse/setup.py.disabled
* helpers/macropad/setup.py.disabled
* helpers/magtag/setup.py.disabled
* helpers/matrixportal/setup.py.disabled

## Do push to PyPI - Blinka can make use of it
* drivers/displayio_ssd1306/setup.py.disabled
* drivers/ov2640/setup.py.disabled
* drivers/ssd1322/setup.py.disabled
* drivers/ssd1327/setup.py.disabled
* drivers/ssd1608/setup.py.disabled
* drivers/ssd1680/setup.py.disabled
* drivers/ssd1681/setup.py.disabled

## Maybe - Could Blinka make use of it?
* drivers/nunchuk/setup.py.disabled
* helpers/boardtest/setup.py.disabled
* helpers/colorsys/setup.py.disabled
* helpers/dash_display/setup.py.disabled
* helpers/itertools/setup.py.disabled
* helpers/logging/setup.py.disabled
* helpers/pybadger/setup.py.disabled
* helpers/pyoa/setup.py.disabled

@makermelissa , @ladyada for input.

If it's agreed to push some of these to PyPI, I intend to work through the selected list and make the changes.

ladyada, 2021-08-09:
i defer to @kattni - i think at least nunchuk should work fine in blinka, dunno how we feel about the rest

lesamouraipourpre, 2021-08-12:
Hi @jepler 
With respect to the OV2640 and OV7670 libraries, I see that there is commit stating that you don't want these uploaded to PyPI. Is there a likelihood that they could be usable from Blinka?

makermelissa, 2021-08-12:
For the SSD libraries, Blinka displayio does not currently have code for making use of these drivers, but it could potentially and if you would like to start adding code, that would be awesome.

lesamouraipourpre, 2021-08-12:
> For the SSD libraries, Blinka displayio does not currently have code for making use of these drivers, but it could potentially and if you would like to start adding code, that would be awesome.

I think all of the displays I have on hand use SH11.. drivers but I've been wanting to get an eInk display so I'll make sure I get one with a SSD.... driver

jepler, 2021-08-12:
>  With respect to the OV2640 and OV7670 libraries, I see that there is commit stating that you don't want these uploaded to PyPI. Is there a likelihood that they could be usable from Blinka?

No, because Blinka does not implement the ParallelCapture class.

jepler, 2021-08-12:
> Is the `imagecapture` core library worth porting to Blinka?

I don't know enough about what would be involved to say for sure. On Linux SBCs, it's already possible to interface with cameras that use the standard V4L APIs, possibly including the OV2640.  (I only say that because this is a source file that exists: https://github.com/torvalds/linux/blob/master/drivers/media/i2c/ov2640.c) But that wouldn't require or benefit from the code in the OV2640/7670 libraries.

kattni, 2021-08-13:
Hey, folks. A few things.

- Anything that exists in Python is not being pushed to PyPI, e.g. itertools, logging. We needed a slimmed down version for CircuitPython, but the full versions would be available on Linux, so it doesn't make sense to include them.
- Boardtest I'm not sure about. I don't know if it'll work properly or not. @lesamouraipourpre If you want to test it, I'm fine with pushing it to PyPI, but without testing, it needs to stay disabled.
- PyBadger is a board-specific helper. There is no reason to push any of the board-specific helper libs to PyPI.
- Dash_Display is currently only designed to work with the FunHouse display. If in the future it is made display-agnostic, we can consider pushing it to PyPI. Until that time, it should remain disabled.
- Nunchuk can be updated to deploy to PyPI.
- PYOA should be tested with Blinka, but if it works, can also be deployed to PyPI.

Please ensure that you are updating the README to have the PyPI install instructions for libraries being deployed to PyPI.

tekktrik, 2022-08-25:
@jepler I think this is now resolved since everything is on PyPI.  Also, I have a [PR out for the cookiecutter](https://github.com/adafruit/cookiecutter-adafruit-circuitpython/pull/200) to remove the option NOT to add it to PyPI.

kattni, 2022-08-25:
Yes this is completed. Thank you for catching it.

closed 2022-08-25 by kattni