adafruit/Adafruit_CircuitPython_Bundle #353: Had to install setuptools_scm as a build dependency

weskerfoot, 2021-11-06:
I got this error while doing `pip install circuitpython-build-tools` in the venv.

```
WARNING: Discarding https://files.pythonhosted.org/packages/f7/a7/62a98a7db8ad85e758802dddfad1d6f3093128d2c7487059629ffcc5d1e6/circuitpython-build-tools-1.8.6.tar.gz#sha256=cc9c33c91a94fff37b8e82bd33e89ff3a3eac3739c343746688583b160b0943e (from https://pypi.org/simple/circuitpython-build-tools/) (requires-python:>=3.7). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Using cached circuitpython-build-tools-1.8.5.tar.gz (318 kB)
    ERROR: Command errored out with exit status 1:
     command: /home/wes/Adafruit_CircuitPython_Bundle/venv/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-b1amccly/circuitpython-build-tools_5723cae64856498cbf8c8c9c344749a1/setup.py'"'"'; __file__='"'"'/tmp/pip-install-b1amccly/circuitpython-build-tools_5723cae64856498cbf8c8c9c344749a1/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-qu21gqia
         cwd: /tmp/pip-install-b1amccly/circuitpython-build-tools_5723cae64856498cbf8c8c9c344749a1/
    Complete output (37 lines):
    WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object at 0x7f286e3a4fa0>: Failed to establish a new connection: [Errno 111] Connection refused')': /alice/dev/+simple/setuptools-scm/
    WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object at 0x7f286e3a4400>: Failed to establish a new connection: [Errno 111] Connection refused')': /alice/dev/+simple/setuptools-scm/
    WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object at 0x7f286e3a42b0>: Failed to establish a new connection: [Errno 111] Connection refused')': /alice/dev/+simple/setuptools-scm/
    WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object at 0x7f286e349310>: Failed to establish a new connection: [Errno 111] Connection refused')': /alice/dev/+simple/setuptools-scm/
    WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object at 0x7f286e349520>: Failed to establish a new connection: [Errno 111] Connection refused')': /alice/dev/+simple/setuptools-scm/
    ERROR: Could not find a version that satisfies the requirement setuptools_scm (from versions: none)
    ERROR: No matching distribution found for setuptools_scm
    Traceback (most recent call last):
      File "/home/wes/Adafruit_CircuitPython_Bundle/venv/lib/python3.9/site-packages/setuptools/installer.py", line 75, in fetch_build_egg
        subprocess.check_call(cmd)
      File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/home/wes/Adafruit_CircuitPython_Bundle/venv/bin/python', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmp784rl7yw', '--quiet', '--index-url', 'http://localhost:3141/alice/dev/+simple/', 'setuptools_scm']' returned non-zero exit status 1.
    ```

For some reason doing `pip install setuptools_scm` fixed it. Looks like that's because of this line https://github.com/adafruit/circuitpython-build-tools/blob/80489a614228d2f34bb8d8b55970df3e875b9b76/setup.py#L7

Might be good to add a note about this?

jepler, 2021-11-06:
> '--index-url', 'http://localhost:3141/alice/dev/+simple/'

It looks like you may have some local configuration (possibly an environment variable called PIP_INDEX_URL) which may be interfering with the `pip` command.

weskerfoot, 2021-11-06:
ah yes you're right, I had totally forgotten about that

closed 2021-11-08 by jepler