adafruit/Adafruit_CircuitPython_Bundle #360: Provide an official way for tools to enumerate supported bundle versions

jepler, 2021-12-16:
circup and possibly the vs code extension both encountered problems when we dropped the 6.x bundles. circup in particular hard coded the list of versions. if we provide an official way to enumerate supported bundle versions for the latest bundle, it will allow these consumers of the bundle to act appropriately when we add or remove a version.

FlantasticDan, 2021-12-19:
Seems like bundles are built and released with the [CircuitPython Build Tools](https://github.com/adafruit/circuitpython-build-tools) package.  Perhaps pointing users to [this file](https://github.com/adafruit/circuitpython-build-tools/blob/main/circuitpython_build_tools/target_versions.py) which should be a stable URL and isn't too difficult to parse is a viable/simple option.

dhalbert, 2021-12-19:
I propose here: https://github.com/adafruit/circup/issues/133 to make circup immune to bundle version changes.