adafruit/Adafruit_CircuitPython_Bundle #370: Error coming while trying to import board

thedim-witted, 2022-03-02:
Hi,

I went through the installation process and was able to detect the ft232h board using the ftdi library.

```
>>> from pyftdi.ftdi import Ftdi
>>> Ftdi().open_from_url('ftdi:///?')
Available interfaces:
  ftdi://ftdi:232h:1/1   (￿￿￿￿￿￿)
```

 However, it throws an error when I try to do a 'import board'
The error says:

>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/devop1/.local/lib/python3.6/site-packages/board.py", line 39, in <module>
    import adafruit_platformdetect.constants.boards as ap_board
  File "/home/devop1/.local/lib/python3.6/site-packages/adafruit_platformdetect/__init__.py", line 10, in <module>
    from .board import Board
  File "/home/devop1/.local/lib/python3.6/site-packages/adafruit_platformdetect/board.py", line 24
    from __future__ import annotations
    ^
SyntaxError: future feature annotations is not defined

FoamyGuy, 2022-03-02:
I believe using Blinka on linux now requires python 3.7 at least. 

Can you try updating to python 3.7 or make a seperate virtual environment on 3.7 and try with that.

thedim-witted, 2022-03-03:
That system has python 3.8.10

FoamyGuy, 2022-03-03:
In the error message that you posted it shows that it is using `site-packages` folder inside of `python3.6` 

Perhaps you have 3.8.10 installed but it appears the code you are attempting to run is running under python3.6

Maybe try removing python3.6 entirely if you can to ensure that your script is executing with the python 3.8.10 instance.

thedim-witted, 2022-03-07:
Hey, so I figured out that it was actually happening because it was running in the python3.6
Unfortunately, as we can't update our python version because of dependency, I downgraded the Adafruit_PlatformDetect to a version that works with python3.6

Thanks for your help though. :)

closed 2022-03-07 by thedim-witted