adafruit/Adafruit_CircuitPython_Bundle #384: Latest bundle version's mpy files aren't compiled.

domdfcoding, 2022-06-17:
In the latest bundle ([June 17](https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20220617) at the time of writing) the `*.mpy` files are actually plain ol' Python files with a `.mpy`extension. I'm not sure they're supposed to be like that. In the previous bundle from the 14th they're binary files. 

This applies to both the 7.x and 8.x bundles.

Idirianuk, 2022-06-17:
May be associated.
CPY 7.3. One library giving me this error:

```
Traceback (most recent call last):
  File "code.py", line 7, in <module>
  File "adafruit_esp32spi/adafruit_esp32spi_wifimanager.py", line 18, in <module>
MpyError: Incompatible .mpy file. Please update all .mpy files. See http://adafru.it/mpy-update for more info.

Code done running.
```
Updated all other project libraries at the same time. Rest are fine.

meltdown03, 2022-06-18:
It seems like it's only the `mpy` files not in a directory, such as `adafruit_ili9341.mpy`, are not compiled. Libraries with their own directory such as `adafruit_rsa` are compiled. Maybe something changed since adding version 8.x to the releases.

Idirianuk, 2022-06-18:
> May be associated. CPY 7.3. One library giving me this error:
>
> ```
> Traceback (most recent call last):
>   File "code.py", line 7, in <module>
>   File [...]

Worked out that this is due to `adafruit_requests.mpy` not being compiled.

Neradoc, 2022-06-19:
The latest release of the Adafruit Bundle is fixed, the Community Bundle will be soon (it isn't yet because there was no update to it's modules to justify the auto release today). Thanks for the reports.

closed 2022-06-19 by Neradoc