adafruit/Adafruit_CircuitPython_Bundle #407: Many libraries with bad circup instructions

Neradoc, 2022-10-16:
Many of the libraries have bad circup instructions, missing the `adafruit_` prefix, or using an incorrect name.
This is related to https://github.com/adafruit/cookiecutter-adafruit-circuitpython/issues/207 but I wanted to list the libraries that need fixing here:
```
libraries/drivers/ltr329_ltr303/README.rst
87:    circup install ltr329-ltr303

libraries/drivers/tsc2007/README.rst
88:    circup install tsc2007

libraries/drivers/ags02ma/README.rst
86:    circup install ags02ma

libraries/drivers/tca8418/README.rst
89:    circup install tca8418

libraries/drivers/tt21100/README.rst
84:    circup install tt21100

libraries/drivers/mmc56x3/README.rst
87:    circup install mmc56x3

libraries/drivers/vl53l4cd/README.rst
87:    circup install vl53l4cd

libraries/drivers/guvx-i2c/README.rst
89:    circup install guvx_i2c

libraries/drivers/uc8151d/README.rst
87:    circup install uc8151d

libraries/drivers/ens160/README.rst
87:    circup install ens160

libraries/drivers/vl53l1x/README.rst
88:    circup install vl53l1x

libraries/drivers/max1704x/README.rst
90:    circup install max1704x

libraries/drivers/nunchuk/README.rst
87:    circup install nunchuk

libraries/drivers/ov5640/README.rst
42:    circup install ov5640

libraries/drivers/scd4x/README.rst
87:    circup install scd4x

libraries/drivers/floppy/README.rst
56:    circup install floppy

libraries/drivers/pcf8575/README.rst
86:    circup install pcf8575

libraries/drivers/pcf8574/README.rst
88:    circup install pcf8574

libraries/drivers/adxl37x/README.rst
88:    circup install adxl37x

libraries/helpers/radial_controller/README.rst
82:    circup install radial_controller

libraries/helpers/ducky/README.rst
87:    circup install ducky

libraries/helpers/ble_beacon/README.rst
83:    circup install ble-beacon

libraries/helpers/pixelbuf/README.rst
82:    circup install pixelbuf
```
Which if I didn't mess up my grep is those repositories:

https://github.com/adafruit/Adafruit_CircuitPython_ltr329_ltr303
https://github.com/adafruit/Adafruit_CircuitPython_tsc2007
https://github.com/adafruit/Adafruit_CircuitPython_ags02ma
https://github.com/adafruit/Adafruit_CircuitPython_tca8418
https://github.com/adafruit/Adafruit_CircuitPython_tt21100
https://github.com/adafruit/Adafruit_CircuitPython_mmc56x3
https://github.com/adafruit/Adafruit_CircuitPython_vl53l4cd
https://github.com/adafruit/Adafruit_CircuitPython_guvx_i2c
https://github.com/adafruit/Adafruit_CircuitPython_uc8151d
https://github.com/adafruit/Adafruit_CircuitPython_ens160
https://github.com/adafruit/Adafruit_CircuitPython_vl53l1x
https://github.com/adafruit/Adafruit_CircuitPython_max1704x
https://github.com/adafruit/Adafruit_CircuitPython_nunchuk
https://github.com/adafruit/Adafruit_CircuitPython_ov5640
https://github.com/adafruit/Adafruit_CircuitPython_scd4x
https://github.com/adafruit/Adafruit_CircuitPython_floppy
https://github.com/adafruit/Adafruit_CircuitPython_pcf8575
https://github.com/adafruit/Adafruit_CircuitPython_pcf8574
https://github.com/adafruit/Adafruit_CircuitPython_adxl37x
https://github.com/adafruit/Adafruit_CircuitPython_radial_controller
https://github.com/adafruit/Adafruit_CircuitPython_ducky
https://github.com/adafruit/Adafruit_CircuitPython_ble-beacon
https://github.com/adafruit/Adafruit_CircuitPython_pixelbuf

tekktrik, 2022-12-07:
Thanks!  This makes fixing them easier!

Note:  Edited your comment to fix the BLE Beacon link.

@kattni - let me know if you want to me to fix this or hand it off to someone else!

kattni, 2022-12-07:
@tekktrik Feel free. I don't have anyone in mind to hand it off to.

tekktrik, 2024-02-26:
@FoamyGuy thanks for the PRs, I removed the auto-linking for them so we wouldn't have to reopen the issue before it was fully fixed.  Is this now done?

FoamyGuy, 2024-02-26:
@tekktrik not quite finished yet, still need to check on the last few. 

I started at the bottom of the list and am working my way up. Some of them seemed to have been fixed in the interim because they already had the circup name notation.

I can close this once I've made it through the rest of them.

FoamyGuy, 2024-11-06:
I found two more of these with a quick check and submitted PRs for them. I believe this is good to be closed now.

closed 2024-11-06 by FoamyGuy