adafruit/Adafruit_CircuitPython_Bundle #432: Bundle being rebuilt with no library changes

dhalbert, 2023-07-03:
Last two nightly bundle builds did not have any library changes. I can't remember: is the bundle rebuilt no matter what? The cron job is updating other things, but if there are no changes, should it still rebuild the bundle? Maybe this is new due to recent CI changes, or maybe I am just not remembering.

Neradoc, 2023-07-03:
It really shouldn't, it seems it has been happening since 10 or 11 Dec 2022:
https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20221211
It has no "updated libraries" or "new libraries" section, and the following release seemingly has 100% identical content.

tekktrik, 2023-07-05:
I'll check on this tonight!

tekktrik, 2023-07-06:
Looking at the relevant code and the timeframe, this lines up with the change I made to publish for auto-tallying library downloads.  It looks like because it makes a commit and push for that markdown page, it causes a release to happen now since "updates" were made.  I can work on changing that, if having what is now effectively a daily bundle release is undesirable.

dhalbert, 2023-07-06:
I think it would be good to suppress releases when there are no changes. It reduces needless downloads, both automatic by `circup`, and manual by people who think something might have changed. Thanks for the diagnosis~

closed 2023-07-11 by tekktrik