adafruit/Adafruit_CircuitPython_Bundle #437: key stuck with macropad

SpudGunMan, 2023-09-19:
The following code block has issues with functionality. 

Expected behavior: all keys work the same, where I could programmatically set key send rate
Possible Issue: Some keys repeatedly send characters and some dont see example code block 

the code will send keys when pressed for example 'tttttttttttttgrrrrrrrrrfbbbbbbbbv' looking at the code this is unexplained.

```
from adafruit_macropad import MacroPad
import supervisor
import time
macropad = MacroPad() 
while True:
    key_event = macropad.keys.events.get()
    if key_event:
        if key_event.pressed:
            if key_event.key_number == 0:
                macropad.keyboard.send(macropad.Keycode.G)
                #single key sent and it stops the other runaway keypresses

            if key_event.key_number == 2:
                macropad.keyboard.press(macropad.Keycode.T)
                #repeats keypresses
                
            if key_event.key_number == 3:
                macropad.keyboard.send(macropad.Keycode.F)
                #single key sent and it stops the other runaway keypresses

            if key_event.key_number == 5:
                macropad.keyboard.press(macropad.Keycode.R)
                #repeats keypresses

            if key_event.key_number == 6:
                macropad.keyboard.send(macropad.Keycode.V)
                #single key sent and it stops the other runaway keypresses

            if key_event.key_number == 8:
                macropad.keyboard.press(macropad.Keycode.B)
                #repeats keypresses
```

caternuson, 2023-09-19:
send() vs. press()

There are two events with each key stroke: (1) press and (2) release. Both of these events can be generated separately by their respective functions. If you call press(), but don't call release(), that is equivalent to holding a key down. So that key value will repeat until release() is called. (appear stuck)

The send() function is a convenience function that combines both a call to press() and send() into one function.

dhalbert, 2023-09-19:
Note that the repeating is done by the host computer, based on how it does "auto repeat". The keyboard (a physical keyboard, or CircuitPython) itself is not sending multiple presses.

SpudGunMan, 2023-09-19:
I am a goof sorry

closed 2023-09-19 by SpudGunMan

dhalbert, 2023-09-19:
np, it's all a learning experience!