adafruit/Adafruit_CircuitPython_Bundle #450: Source code release zip is mostly empty directories.

clickonben, 2023-12-17:
The source code .zip and .tar.gz files that get produced as part of each release have an empty directory structure in the "libraries" directory, there is no code at all. It looks like whatever builds these doesn't properly process the git submodules so anything that is a submodule is just an empty directory in both zips.

FoamyGuy, 2023-12-17:
This is the link to use for the python source code for all libraries in the bundle
[image]

It looks populated correctly as far as I can tell. I don't believe the links below are intended to have things in them, but I'm not certain.

clickonben, 2023-12-17:
So what you are saying is that the files named "source code" don't contain much if any source code and are supposed to be mostly just empty directories? 

Assuming there is some purpose to these files that escapes me, I would politely suggest that they are renamed or at least an explanation is added to the download page that the archive files that are labelled "source code" aren't the ones that contain the source code.

FoamyGuy, 2023-12-17:
I think the links for Source Code zip and tar.gz are created automatically by Github and it seems that they are known to not populate code for submodules based on the discussion here: https://github.com/softprops/action-gh-release/issues/220

I didn't dig too deep but it after a quick look it doesn't seem possible to disable them or rename the links to something different unfortunately. Perhaps we can try to come up with some clarifying language to add to the release notes.

jepler, 2023-12-18:
I want to confirm what @FoamyGuy says -- as far as I know there is no way to remove or suppress the "source code (zip)" or "source code (tar.gz)" links that are automatically added by github, and there's no way to make them include the content of the files that are actually used to build the bundle.

If there's something actionable about this for us, it could be to add a note in the standard release notes that says that due to github technical limitations those links are not useful. Otherwise, there does not seem to be much that we can do.

dhalbert, 2023-12-18:
I was looking for the standard release notes template, but could not immediately find where they came from, even with GitHub organization search. I agree that's what's to be done here. I did something similar in the release notes for `adafruit/circuitpython`.

jepler, 2023-12-18:
It looks like the text we use comes from adabot
```
adabot/circuitpython_bundle.py:        " Please download the one that matches the major version of your CircuitPython. For example"
```

closed 2023-12-18 by dhalbert