adafruit/Adafruit_CircuitPython_Bundle #464: Request Inclusion of .MPY Files in Github Repository

JohnHind, 2024-06-12:
For simple libraries with just .py files I find by far the most comfortable workflow is just to search for a library on Github and download the 'raw' .py file directly to the lib folder on the CIRCUITPY drive. But this workflow fails if the bundled library has the files compiled to .mpy. Everything would be so much easier if the compiled .mpy files were in the repository along with the source files. Rather than checking the .mpy files into the repository, it might be better to use the 'Releases' mechanism to make it downloadable as a zip file, essentially the same as what is in the bundle, but just for that one library.

A search on a device name on Git will usually yield the right repository immediately regardless of which bundle it is in, whether is is a .py file, a .mpy file or a library folder. It will even usually find it if it is an unofficial driver that has not been adopted. Unlike Circup, you do not need to know the precise name of the library. Also you get all the information you need in one place and do not have to hunt it down, particularly in the truly horrible 'read the docs' site! I often find myself going to Github first and then having to spend painfully unnecessary time tracking the compiled version down in the bundles. The bundle zips now take quite a long time to download and unpack which is not optimal just to find a few library updates.

dhalbert, 2024-06-12:
> Rather than checking the .mpy files into the repository, it might be better to use the 'Releases' mechanism to make it downloadable as a zip file, essentially the same as what is in the bundle, but just for that one library.

If I understand you correctly, we do have these .zip files already, for example: https://github.com/adafruit/Adafruit_CircuitPython_HID/releases/tag/6.1.1.

> Unlike Circup, you do not need to know the precise name of the library.

circup has a "match" feature: https://github.com/adafruit/circup/pull/62

> Are you talking about https://docs.circuitpython.org/projects/bundle/en/latest/drivers.html ?

I opened an issue in the past about including the library name in there: https://github.com/adafruit/Adafruit_CircuitPython_Bundle/issues/418

JohnHind, 2024-06-12:
My apologies Dan! You know how sometimes you are looking right at something and not seeing it! I was expecting to see 'Releases' in the menu bar along the top. I think this does what I wanted. It might be better for the Learn Guides to point people to this and/or Circup. The bundles are getting way too large for comfort!

closed 2024-06-12 by JohnHind

dhalbert, 2024-06-12:
The bundles are 13-14MB each at this point. If you have slow internet, I can see that might be a bit annoying, though it's basically once or day or less.

I made a typo in my reply (now fixed) and didn't make clear my question about the readthedocs page for libraries. If you have some suggestions, let us know.

JohnHind, 2024-06-12:
The size does also make it hard to find the right driver, especially as it might be a folder, a py or a mpy file, and it might be in one of two different bundles (and sometimes not in a bundle at all).

My acid comment about ReadTheDocs was this sort of thing:
https://docs.circuitpython.org/projects/ch9328/en/latest/
Versus this, which I think this is a much better experience:
https://github.com/adafruit/Adafruit_CircuitPython_CH9328
Including the experience of navigating to these pages in the first place when all you know is the chip reference CH9328.

But many thanks for your help. I am writing my first 'Playground' article which is about developing for Adafruit products using VS Code for a range of scenarios, Arduino, CircuitPython (USB Serial and UPDI) and Blinka on a remote Raspberry Pi.