adafruit/Adafruit_CircuitPython_Bundle #487: Xiao 2350

RMarcelPipa, 2024-11-27:
Someone can create te configuration from Xiao Seeed 2350 ?

closed 2024-11-27 by tannewt

RMarcelPipa, 2024-11-27:
yes I know about this... I need the profile for Vs Code for the autocomplete

RMarcelPipa, 2024-11-27:
I Did the dir on Board Module and I found this.

`['__class__', '__name__', 'A0', 'A1', 'A2', 'A3', 'BAT_ADC_EN', 'D0', 'D1', 'D10', 'D11', 'D12', 'D13', 'D14', 'D15', 'D16', 'D17', 'D18', 'D2', 'D3', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'I2C', 'LED', 'MISO', 'MOSI', 'NEOPIXEL', 'NEOPIXEL_POWER', 'RX', 'SCK', 'SCL', 'SDA', 'SPI', 'TX', 'UART', 'VOLTAGE_MONITOR', '__dict__', 'board_id']`

FoamyGuy, 2024-11-27:
@RMarcelPipa To get the stubs and board pin listing working for you device you need to install `circuitpython-stubs` and use a command to set which board you want to have the pins for. 

You should be able to install the stubs with:
```
pip install circuitpython-stubs
```
Then set your board using the `circuitpython_setboard` command. [This guide page](https://learn.adafruit.com/welcome-to-circuitpython/pycharm-and-circuitpython#device-specific-board-stubs-3168661) details the process. It's written for PyCharm, but should apply similarly for VSCode.